cameroncros/OctoPrint-DiscordRemote #232: 2 requests in one ask: Multiple servers support. Multiple command characters support.

DrToxic, 2023-01-03:
**Is your feature request related to a problem? Please describe.**
_Multiple Servers Support_
I run bots on my discord. a friend of mine would like to use his bot on both mine and his servers, partly so I can watch and debug.

_Multiple Command Characters Support_
It'd be pretty dope to see all bots respond at the same time to trivial commands, like !Status, however it would be irritating to, for example, start all printers printing the same file.

So have each printer reacting to it's own command character (like it is now), but also assign a second character for trivial commands, such as /Status.

**Describe alternatives you've considered**
Running multiple plugin instances, setting up multiple bots per printer.

**Additional context**
[image]

cameroncros, 2023-01-04:
Would being able to DM commands to the bots be sufficient? Its probably the easiest solution that doesnt involve a very convoluted configuration?

I'll look into the ability to DM the bots.

cameroncros, 2023-01-04:
See v3.11, allows DM'ing the bot directly. With that, you can have your bots on the same channel, and send commands to all bots, or use DM for individual commands. I will think about the multiple channels thing, but I cant promise any kind of prompt fix for that

DrToxic, 2023-01-04:
That could work nicely. My only thought is, I have 3 printers and want to
keep tabs on them all.

I'd like to extend a great thank you for developing a great, seriously
useful plugin.

Dr. Toxic

DrToxic, 2023-01-08:
> See v3.11, allows DM'ing the bot directly. With that, you can have your bots on the same channel, and send commands to all bots, or use DM for individual commands.

Amazing stuff.

> I will think about the multiple channels thing, but I cant promise any kind of prompt fix for that
I will take a look at the source and see if I can't work anything out. I just happen to be busy as heck at the moment. :(

Liftyee, 2023-01-21:
I had made an attempt at making it work for multiple channels, but it requires a bit of an overhaul to only send some responses to one channel (otherwise there are out of context responses in other channels), or to show who ran what command where in the response. I also no longer needed to have it run in multiple channels, though I could try and continue working on this feature if people want it.

anmart, 2023-04-02:
Would it be possible to have the command responses go to the person that DM'd it or would that require adding extra channel IDs? I DM'd my bot the `snapshot` command and it posted an image to the main channel it's present in

cameroncros, 2023-04-03:
That shouldn't be too hard to do, and if you want to make that change I
will definitely merge it in.

Ideally would want to keep the existing "send to main channel" behaviour
though as someone may already be using that feature.