cameroncros/OctoPrint-DiscordRemote #233: Updating to 3.11.0 causes discordremote to no longer show up as installed and unable to be configured.

bharvey88, 2023-01-08:
**Describe the bug**
Once I updated discordremote to 3.11.0 and restarted I noticed my discord bot was offline. I restarted the pi and it still wasnt back so I went into the octoprint ui and discordremote didnt show as installed. So I then reinstalled, then restarted, and still the plugin doesnt show installed. https://imgur.com/6Zs17Ug here is a picture showing octoprint shows "unknown" was installed successfully in top right.

*Logs*
Please find the logs and upload them here. It'll likely be the first thing that is asked for, so save some time and upload them. Logs can be found under Settings -> Logging -> plugin_discordremote.log

the only logs found in above location are from before the update. possibly due to octoprint not properly seeing the plugin installed correctly.

**To Reproduce**
Steps to reproduce the behavior:
1. update discord remote
2. restart octoprint
3. discord remote is now not showing up as installed and unable to be configured.

**Expected behavior**
Update succeeds and plugin still works.

**Screenshots**
https://imgur.com/6Zs17Ug here is a picture showing octoprint shows "unknown" was installed successfully in top right.

**Desktop (please complete the following information):**
 - OS: windows 11
 - Browser firefox
 - Version latest stable

**Additional context**
none needed

bharvey88, 2023-01-08:
follow up - I spoke to jneilliii (octoprint / plugin dev) and he suggested trying this command via cli "/home/pi/oprint/bin/pip install --force-reinstall --no-cache-dir https://github.com/cameroncros/OctoPrint-DiscordRemote/archive/refs/tags/v3.11.0.zip" and after rebooting the plugin is now showing back up and working.

bharvey88, 2023-01-08:
second follow up - succeeded upgrading second printer plugin with no issues. Possibly a corruption issue on sd card. I wont close this just in case you think it might be worth looking into, but if you agree it's most likely just a corruption issue then please feel free to close it.

cameroncros, 2023-02-02:
Im happy to close the issue, If I cannot replicate it then I cant really debug it, and it seems like you solved it for yourself anyway.

closed 2023-02-02 by cameroncros