cameroncros/OctoPrint-DiscordRemote #234: Send a notification when <X> layer starts.

McStebb, 2023-01-11:
**Is your feature request related to a problem? Please describe.**
No.

**Describe the solution you'd like**
Sometimes I print multi-color models by manually pausing and changing filament at a specific layer, but it usually involves me babysitting the print until I can visually tell it's at the right layer.

It would be nice if Octoprint had a setting that sent a notification (with optional webcam snapshot) when the print reached <X> layer. That way I can go pause and change filament when I get notified.

Most slicers append a layer number comment line at the start of each layer, so triggering the bot to send a notification could simply look for ";LAYER <X>" for when to send the notification.

Or just have the user add custom comment gcode at layer via the slicer. Something like this.
(Add custom GCode at layer 173...)
" ;NOTIFY "

And then the bot notifies me at that layer because of the " ;NOTIFY " comment.

**Describe alternatives you've considered**
I could just install a multi-filament mod, but that's a considerable undertaking.

**Additional context**
N/A

rtdog, 2025-02-20:
For example in PrusaSlicer I could add an "After layer change G-code" setting `{if layer_num == 2 } M117 INDICATOR-Layer2 {endif} ; ` and I would think your plugin could capture this message for action.

rtdog, 2025-02-20:
Ok, sorry for the SPAM.  It appears you already support "specific gcode sent", so I think this is already implemented!

rtdog, 2025-02-20:
Ok I implemented with M240 on Layer 2 using the "specific gcode" notification feature and it worked great, I just stuff an M240 in my gcode on layer 2 and I'm good.  M240 is "Trigger Camera" and Prusa ignores it, so that works well enough.

cameroncros, 2025-02-20:
So its working then? Excellent. @McStebb does the gcode solution described work for you? Should I close off the issue?

McStebb, 2025-02-20:
> For example in PrusaSlicer I could add an "After layer change G-code" setting `{if layer_num == 2 } M117 INDICATOR-Layer2 {endif} ; ` and I would think your plugin could capture this message for action.

I tried this and it worked!  Thanks!
(I had forgotten this post - sorry for leaving it open so long!)

cameroncros, 2025-02-20:
Closing as complete, thanks rtdog and mcstebb

closed 2025-02-20 by cameroncros