cameroncros/OctoPrint-DiscordRemote #237: Dependency issues with OctoPrint 1.9.0rc2

MaleNurse, 2023-03-16:
**Describe the bug**
In order to upgrade to the latest OctoPrint RC, I had to uninstall DiscordRemote due to at least one dependency conflict:
aiohttp 3.7.4.post0 requires async-timeout<4.0,>=3.0, but you have async-timeout 4.0.2 which is incompatible.

EvilGremlin, 2023-04-02:
Issue is even worse. attempt to install plugin on RC3 result in unconditional revert to 1.8.7

cameroncros, 2023-04-17:
This is out of my control, I dont set the async-timeout dependency, it is pulled in by discord.py.

EvilGremlin, 2023-04-17:
you can fork and tweak it at least as temporary fix

cameroncros, 2023-04-17:
Are you sure its octoprints dependency? Not another plugin? I cant seem to get it to fail to install for me, and I am testing against octoprint dev

MaleNurse, 2023-04-18:
pipdeptree shows 

OctoPrint==1.9.0rc5
    zeroconf==0.39.4
        async-timeout==4.0.2`

EvilGremlin, 2023-04-18:
No, after installing rc3 discorderemote disappeared, then installing only it reverted octo to 1.8.7
Updated to rc5, WTF if happening now? This can't be right.
[log.txt](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/11258197/log.txt)

cameroncros, 2023-04-18:
What version of python are you on? If I upgrade to discord.py==2.2.2, then everything will work, but it wont support python3.7 and lower. Given python3.7 is about to EOL in a few months, that seems like a reasonable compromise?

cameroncros, 2023-04-19:
Octopi is python3.7 though, so that will cause problems.

EvilGremlin, 2023-04-19:
3.10.6

cameroncros, 2023-04-19:
Release v4.0.0, should maybe work now for you? I have an old octopi
install, so going through a bit of update pain.

EvilGremlin, 2023-04-19:
this worked :+1:

closed 2023-04-19 by cameroncros

MaleNurse, 2023-04-20:
Worked for me as well. Thank you!