cameroncros/OctoPrint-DiscordRemote #242: Bot Permissions Issue

luccassa, 2023-04-23:
I can only get the bot to greet, no commands can be issued from both discord or the octoprint client. I followed the issue #226 and made sure an asterix was present on commands and users. I also made sure the bot had all three settings enabled and it had full admin privilege. The bot connects, sends wake-up text, snapshot of my printer, then is unresponsive.

[image]
[image]

Austnikl, 2023-04-23:
Hey are you able to send a command through Octopi i seem to have a similar issue. 

Are you able to send a status command from octopi?

luccassa, 2023-04-23:
No commands are accepted either from the Octoprint UI, or discord.

Austnikl, 2023-04-23:
i am having the same issue im able to get this msg and nothing else . 

[image]

here are my logs 
[plugin_discordremote.log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/11305237/plugin_discordremote.log)
[plugin_pluginmanager_console.log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/11305238/plugin_pluginmanager_console.log)
[octoprint.log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/11305239/octoprint.log)

Austnikl, 2023-04-24:
i have enabled the privileged gateways 

[image]

luccassa, 2023-04-24:
> I have enabled the privileged gateways. and still can't get any msg sent across.

Same here

cameroncros, 2023-04-24:
@Austnikl Do you have Octorant and DiscordRemote installed at the same time? AFAIK they shouldnt conflict, but I have never tested it.

cameroncros, 2023-04-24:
Are you both using v4.1 of DiscordRemote?

Austnikl, 2023-04-24:
i am however i was having this issue before installing Octorant. and i am running Discordremote 4.1

cameroncros, 2023-04-24:
Try uninstalling v4.X and installing the latest v3.X version:

https://github.com/cameroncros/OctoPrint-DiscordRemote/archive/refs/tags/v3.11.0.zip

Some people have had issues with v4.x, so it may be worth trying the older version until I fix things.

Austnikl, 2023-04-24:
i will give it a go and let you know.

Austnikl, 2023-04-24:
that version you sent me works I can send commands in discord.

luccassa, 2023-04-24:
Worked for me as well. I will say that I tried 4.2 and the commands were
functioning from the UI side only. However, it was extremely slow
compared to the functional 3.11 build. Thanks for the help and awesome
plugin. Good luck!

cameroncros, 2023-04-24:
Sounds like your both happy now, I will close this off. Create a new issue if you have any other problems.

closed 2023-04-24 by cameroncros