cameroncros/OctoPrint-DiscordRemote #243: Error after upgrading to the latest release

talentedmrripley, 2023-04-24:
After updating the plugin today I am getting this error:

`2023-04-24 04:00:43,423 - asyncio - ERROR - Task exception was never retrieved
future: <Task finished name='Task-89' coro=<DiscordImpl.process_message_queue() done, defined at /octoprint/plugins/lib/python3.8/site-packages/octoprint_discordremote/discordimpl.py:137> exception=RuntimeError("Task <Task pending name='Task-89' coro=<DiscordImpl.process_message_queue() running at /octoprint/plugins/lib/python3.8/site-packages/octoprint_discordremote/discordimpl.py:144>> got Future <Future pending> attached to a different loop")>
Traceback (most recent call last):
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint_discordremote/discordimpl.py", line 144, in process_message_queue
    await self.process_queue.wait()
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint_discordremote/discordimpl.py", line 42, in wait
    await self.event.wait()
  File "/usr/local/lib/python3.8/asyncio/locks.py", line 309, in wait
    await fut
RuntimeError: Task <Task pending name='Task-89' coro=<DiscordImpl.process_message_queue() running at /octoprint/plugins/lib/python3.8/site-packages/octoprint_discordremote/discordimpl.py:144>> got Future <Future pending> attached to a different loop`

Several restarts and a clean install did not help.

talentedmrripley, 2023-04-24:
If there's anything else I can get to you to assist let me know, I'm happy to provide it.

LyricPants66133, 2023-04-24:
@talentedmrripley What are your octoprint and python versions?

talentedmrripley, 2023-04-24:
@LyricPants66133 

Python 3.8.16
Octoprint Version 1.8.7

cameroncros, 2023-04-24:
Does release v4.2 work for you?

cameroncros, 2023-04-24:
v4.2 works for me, but there is still a bug in the events, so its very slow, and will take up to 10 seconds to send a message through

talentedmrripley, 2023-04-24:
Yeah, 4.2 is better in some ways.  I am no longer getting the error and I can now send commands via OctoPrint that are working in Discord.  But sending commands from Discord still fails and weirdly when sending a /status from OctoPrint the IP info, both internal and external, is wrong.  

I'll go back to a 3.x release for the time being.  Where can I find them?  I looked under releases but only see source code.

cameroncros, 2023-04-25:
https://github.com/cameroncros/OctoPrint-DiscordRemote/archive/refs/tags/v3.11.0.zip

cameroncros, 2023-04-25:
v4.3 should fix the problems now.

darloxflyer, 2023-04-25:
Sooo, 4.3 actually seems to have made the problem worse.  Was having this same issue as OP after an upgrade to 4.2.  Upgraded to 4.3, and now seem to have no Discord connectivity whatsoever.  Two new errors appearing, copied below.  Same configuration directives on an older version of the plugin still seems to work ok.  

Now seems to constantly drop the following error in octoprint.log with an increasing reconnect wait time each time:

-----------
2023-04-24 21:29:31,437 - octoprint.plugins.discordremote - INFO - Sending msgs
2023-04-24 21:30:24,249 - discord.client - ERROR - Attempting a reconnect in 54.25s
Traceback (most recent call last):
  File "/home/jkilmer/oprint/lib/python3.9/site-packages/discord/client.py", line 659, in connect
    await self.ws.poll_event()
  File "/home/jkilmer/oprint/lib/python3.9/site-packages/discord/gateway.py", line 646, in poll_event
    raise ConnectionClosed(self.socket, shard_id=self.shard_id, code=code) from None
discord.errors.ConnectionClosed: Shard ID None WebSocket closed with 1000
------------

Also the following error is now present where it wasn't before, and the channel_id IS properly set in the DiscordRemote plugin configuration.  (Tried re-saving, but original value was still present from before upgrade)

2023-04-25 14:59:54,446 - octoprint.plugins.discordremote - INFO - DiscordRemote is started !
2023-04-25 14:59:54,463 - octoprint.plugin - ERROR - Error while calling plugin discordremote
Traceback (most recent call last):
  File "/home/jkilmer/oprint/lib/python3.9/site-packages/octoprint/plugin/__init__.py", line 273, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/jkilmer/oprint/lib/python3.9/site-packages/octoprint/util/__init__.py", line 1688, in wrapper
    return f(*args, **kwargs)
  File "/home/jkilmer/oprint/lib/python3.9/site-packages/octoprint_discordremote/__init__.py", line 190, in on_after_startup
    self.configure_discord(False)
  File "/home/jkilmer/oprint/lib/python3.9/site-packages/octoprint_discordremote/__init__.py", line 164, in configure_discord
    self.discord = DiscordImpl(self._settings.get(['bottoken'], merged=True),
  File "/home/jkilmer/oprint/lib/python3.9/site-packages/octoprint_discordremote/discordimpl.py", line 73, in __init__
    self.channel_id = int(channel_id)
ValueError: invalid literal for int() with base 10: ''

darloxflyer, 2023-04-25:
Checked and confirmed.  Definitely not a configuration issue (re-installing 3.11 directly overtop of the 4.3 plugin restores operation as-is without any configuration changes.)

The error messages above are actually inconsistent.  I completely un-installed and re-installed v4.3 fresh, and I no longer get the above errors.  But I don't have any working Discord integration either -- no wake-up message, nothing.  4.2 provides the wake-up message and responds to commands from within the Octoprint UI, but doesn't respond to messages from within the Discord channel.  4.3 is just... nothing.  No errors either, when sending from the Octoprint UI.  It just sits there silently.

Sorry if I sent you off on some wild goose chase with the earlier errors, but I'm guessing they're spurious, or somehow related to the upgrade path followed.  With a fresh install they do not appear, but the plugin is still non-functional. :(

darloxflyer, 2023-04-25:
FWIW, there is another error I just found that's not thrown by DiscordRemote itself, but rather by the new discord.py library that was put into use:

2023-04-25 15:25:26,995 - asyncio - ERROR - Task exception was never retrieved
future: <Task finished name='Task-24' coro=<Client.start() done, defined at /home/jkilmer/oprint/lib/python3.9/site-packages/discord/client.py:756> exception=PrivilegedIntentsRequired("Shard ID None is requesting privileged intents that have not been explicitly enabled in the developer portal. It is recommended to go to https://discord.com/developers/applications/ and explicitly enable the privileged intents within your application's page. If this is not possible, then consider disabling the privileged intents instead.")>
Traceback (most recent call last):
  File "/home/jkilmer/oprint/lib/python3.9/site-packages/discord/client.py", line 778, in start
    await self.connect(reconnect=reconnect)
  File "/home/jkilmer/oprint/lib/python3.9/site-packages/discord/client.py", line 704, in connect
    raise PrivilegedIntentsRequired(exc.shard_id) from None
discord.errors.PrivilegedIntentsRequired: Shard ID None is requesting privileged intents that have not been explicitly enabled in the developer portal. It is recommended to go to https://discord.com/developers/applications/ and explicitly enable the privileged intents within your application's page. If this is not possible, then consider disabling the privileged intents instead.

This is also an interesting one, since in the Discord Developer Portal, I have all of the intrinsic intents enabled:
[image]

There are no EXPLICIT permissions enabled in the more granular Bot Permissions array below that though.  Maybe the new library is looking for some nonstandard permission in this case??  But this could also be spurious or unrelated, as the message only appears in my logs ONCE, even with all the messing around I've been doing...

cameroncros, 2023-04-25:
v4.3 is requesting all intents (Presence, Members and Message Content), which is arguably more than what is required, as Members is not used.

I think your issue may not be plugin related, but may instead be discord/bot related? It is bizarre that one version works and then a minor change will break everything.

Try resetting your bot token and reconfiguring DiscordRemote with the new token, Discord will sometimes disable tokens if there are too many restarts/crashes/disconnects. I have been stung by that twice in the last week, first time I got an email, second time the bot just stopped working.

darloxflyer, 2023-04-25:
Yep, that did it!  Don't ask me why, but it did...

However, that speaks to something potentially very fragile in the upgrade path here, perhaps if people upgraded to 4.0-4.2 in between the last stable at 3.11?  My path was:

* Running 3.11
* Got prompt in Octoprint UI informing me of v4.2
* Updated to 4.2 via Plugin Manager
* Encountered problem described by this ticket - upgraded to 4.3 manually
* Chaos ensues...

So not anything I'd consider to be nonstandard.  So hopefully anyone going 3.11->4.3 will be ok, but the issues with the intermediate versions may have nuked the bot token.  *shrug*

THAT SAID... I see someone else filed a bug indicating Date/Time issues in the Discord messages with 4.3, and can also confirm that's an issue.  See below for back-to-back messages, the first one from 3.11, the immediately-subsequent one from 4.3.

[image]

Anyhow - thanks for your help, and hope not too many others fall into this hole!

cameroncros, 2023-04-25:
I dont have any control over Discord disabling the tokens, I have reached out to them about it, but I dont think they will do anything about it.

I have a fix for the timestamps. I have no idea how/why they have broken.

closed 2023-04-25 by cameroncros