cameroncros/OctoPrint-DiscordRemote #244: Date & Time Mismatch

Mscampbell7, 2023-04-25:
*Description*
I could be absolutely dense, but the date and time are mismatched for me. Possibly a Discord/OctoPrint setting issue?

*Logs*
Attached

**To Reproduce**
Steps to reproduce the behavior:
Standard installation of the plugin, host computer, and receiving computer display correct date and time, automatic sync/timezone

**Expected behavior**
Discord time matching message time

**Screenshots**
[image]

**Desktop:**
 - OS: Windows 10
 - Browser Chrome

**Smartphone:**
 - Device: N/A

[octoprint-logs.zip](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/11315477/octoprint-logs.zip)

cameroncros, 2023-04-25:
Is the time and timezone on your octoprint server correct?

Mscampbell7, 2023-04-25:
The time and timezone on the physical computer running octoprint are correct, is there a separate setting to set the time and timezone of octoprint itself?

cameroncros, 2023-04-25:
No, I dont think so. I just noticed I can replicate as well, my messages are 1 hour out. 

Will try switch from `datetime.utcnow()` to `datetime.now()` and see if that helps.

cameroncros, 2023-04-25:
v4.4 should fix the timestamps

Mscampbell7, 2023-04-26:
I updated to 4.4, and now the bot is stuck as "offline", despite having the right channel ID and a fresh token

cameroncros, 2023-04-26:
😱 There is no way that v4.4 could have broken the connection:

https://github.com/cameroncros/OctoPrint-DiscordRemote/compare/v4.3.0...v4.4.0

It is 3 lines that changed, none of them are critical to the bot functionality. Can you provide the logs from octoprint?

Mscampbell7, 2023-04-26:
[plugin_discordremote.log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/11327830/plugin_discordremote.log)
 
It looks like the Discord Remote is running fine; however, I am getting a separate error now I believe that is caused by octolapse.

"2023-04-24 18:51:03,002 - octolapse.camera - ERROR - Unable to connect to the provided snapshot URL of http://tako3.prefecture/webcam/?action=snapshot for the 'Webcam - Default OctoPi 0.16.0' camera profile."

Do you want the log for just DiscordRemote or all of octolapse?

cameroncros, 2023-04-26:
I don't have anything to do with Octolapse, I suggest you open an issue
with them.

https://github.com/FormerLurker/Octolapse

Mscampbell7, 2023-04-26:
I've fixed the problem with octolapse, looking at the changes I agree there is absolutely no reason the change from 4.3 to 4.4 should have cause this to not work anymore. I'm going to try going back to 4.3 just to double check.

closed 2023-06-17 by cameroncros