cameroncros/OctoPrint-DiscordRemote #245: State that all intents must be selected in README

Coder-Tavi, 2023-04-28:
**Is your feature request related to a problem? Please describe.**
When setting up the bot, I had a lot of trouble and until I opened `octopi.log`, I couldn't find the source of my troubles. Turns out, the program requests intents that the bot account is not cleared for. Due to this, Discord rejected the bot's startup attempt.

**Describe the solution you'd like**
If possible, placing a notice to users that they must enable every single privileged intent would be very useful, especially for first-time users! I've attached a screenshot below of what that looks like.

**Describe alternatives you've considered**
Alternatively, Discord's `Interaction` system could be applied, removing the need for privileged intents (Message Content). I haven't poked into the code a lot but I notice the bot requests all intents, which isn't great if you're getting a bunch of events you do nothing with. You could reduce the intents by restricting it to the `messageCreate` events (I think that's what you only need, but I could be wrong).

**Additional context**
Error in the log file
```
2023-04-28 03:17:15,166 - octoprint.plugins.discordremote - INFO - Received event: startup
2023-04-28 03:17:15,773 - octoprint.plugins.discordremote - INFO - Received event: test
2023-04-28 03:17:16,320 - asyncio - ERROR - Task exception was never retrieved
future: <Task finished name='Task-5990' coro=<Client.start() done, defined at /home/pi/oprint/lib/python3.9/site-packages/discord/client.py:756> exception=PrivilegedIntentsRequired("Shard ID None is requesting privileged intents that have not been explicitly enabled in the developer portal. It is recommended to go to https://discord.com/developers/applications/ and explicitly enable the privileged intents within your application's page. If this is not possible, then consider disabling the privileged intents instead.")>
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.9/site-packages/discord/client.py", line 778, in start
    await self.connect(reconnect=reconnect)
  File "/home/pi/oprint/lib/python3.9/site-packages/discord/client.py", line 704, in connect
    raise PrivilegedIntentsRequired(exc.shard_id) from None
discord.errors.PrivilegedIntentsRequired: Shard ID None is requesting privileged intents that have not been explicitly enabled in the developer portal. It is recommended to go to https://discord.com/developers/applications/ and explicitly enable the privileged intents within your application's page. If this is not possible, then consider disabling the privileged intents instead.
```

[image]
*(And yes, the image for the bot's profile picture is a real image that Discord's made. Check it out [here](https://cdn.discordapp.com/embed/avatars/5.png). I edited it in to redact my bot's info)*

cameroncros, 2023-04-29:
I need more than message create. DiscordRemote listens for messages with commands and responds to them, which. requires the Message Content intent.

If you only use the notification system of discord, you may want to consider Octorant, which is the plugin this grew from, which doesnt even require a bot.

I am happy for you to update the documentation, or even submit an MR to fix whatever issues you find with the permissions, but I have no free time to update and correct it myself.

Coder-Tavi, 2023-05-02:
Good to know!

LoganMD, 2023-05-04:
I second this, this is a requirement for some of the recent updates of DiscordRemote to actually function properly.

Coder-Tavi, 2023-05-18:
Opened a PR that fixes this. While I can't write the code to fix this, I added some extra documentation. Hope this helps!

PR #251 (2023-05-22): fix: add privileged intent message in README
* Points out that users will need to enable privileged intents
* Attaches an image for ease of use

Fixes #245
closed 2023-05-22 by cameroncros