cameroncros/OctoPrint-DiscordRemote #248: Latest two versions don't work at all

EvilGremlin, 2023-05-07:
Just stopped working completely, nothing sent, no response to commands and nothing in the logs. I even recreated bot and it didn't change anything.
Ubuntu 22.04.1, Octoprint 1.9.0rc5, Python 3.10.6
[octoprint-systeminfo-20230507190815.zip](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/11415798/octoprint-systeminfo-20230507190815.zip)
[plugin_discordremote.log](https://github.com/cameroncros/OctoPrint-DiscordRemote/files/11415799/plugin_discordremote.log)

Mscampbell7, 2023-05-08:
I can't seem to figure this one out as well. I thought it was related to the update for corrected date/time, but after talking with @cameroncros and reviewing changelog, I have not the slightest idea why this would cause the bot to stop working. Been stuck offline for me for a while, @EvilGremlin let me know if you can get it working.

cameroncros, 2023-05-08:
Would either of you be comfortable sending me your bot token and channel ID? The plugin is working fine with my bot, so I can test that it works for your bots.

Additionally, can you give me the output of `pip list` on your octoprint venv?

EvilGremlin, 2023-05-08:
np, can you ping me in octo/marlin discord?

cameroncros, 2023-05-08:
So, seems you are both right, and it doesn't work super reliably still. I am gonna try a whole different approach. Rewrite number 4....

Dildacorn, 2023-05-19:
I think I'm going to stick to telegram plugin for now... I thought I was doing something wrong because it says my bot is offline and isn't responding to commands. Is a solution around the corner?

EvilGremlin, 2023-05-19:
it works fine, bit you might have to regenerate token a few times until it'll like one, i think there is issue with special characters somewhere (tokens with double dash or double dots don't work)

cameroncros, 2023-05-20:
I am still investigating, I have no idea why some tokens work and some don't

a1co, 2023-05-21:
Did a first time install and ran into the same problem. Regenerating the tokenb worked for me. But i had to do so 4 times.

cameroncros, 2023-05-22:
If anyone hits this again, please keep a list of all the tokens that failed, I wonder if there is any obvious reason why its not working with some tokens and not others

Mscampbell7, 2023-05-22:
I still haven't found one that works for me, but I'll keep trying and write a list of all the ones that did not work. Do I need to reinvite the bot every token reset? I would assume not.

cameroncros, 2023-05-22:
Nope, just update the token.

cameroncros, 2023-06-17:
@EvilGremlin @Mscampbell7 If you are still interested in using this plugin, try this latest version of the plugin:

https://github.com/cameroncros/OctoPrint-DiscordRemote/archive/refs/heads/cloud-discordshim.zip

This is what the next version of DiscordRemote will be. Instead of setting up the discord bot yourself, everyone will just use my bot, which I am hosting on a server. There are no more python versioning issues with this arrangement, and the discord bot can be self hosted for those who care.

Code for the discord bot: https://github.com/cameroncros/discordshim_rs

cameroncros, 2023-06-29:
5.0.0 is released, which should be a viable solution to this problem. I am going to close off this ticket.

closed 2023-06-29 by cameroncros