cameroncros/OctoPrint-DiscordRemote #252: ERROR: No matching distribution found for discord.py==2.2.2

KenwoodFox, 2023-06-04:
When updating it looks like the plugin cannot find a distribution for discord.py 2.2.2, current latest version is 2.2.3

# Installer Logs
```
Updating, please wait.+++++++++++++++++++++++++++++++++++++++++++Now updating DiscordRemote Plugin to v4.6.0+++++++++++++++++++++++++++++++++++++++++++/home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install https://github.com/cameroncros/OctoPrint-DiscordRemote/archive/v4.6.0.zip --no-cache-dirLooking in indexes: https://pypi.org/simple, https://www.piwheels.org/simpleCollecting https://github.com/cameroncros/OctoPrint-DiscordRemote/archive/v4.6.0.zipDownloading https://github.com/cameroncros/OctoPrint-DiscordRemote/archive/v4.6.0.zipRequirement already satisfied: pillow in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-DiscordRemote==4.6.0) (8.4.0)ERROR: Could not find a version that satisfies the requirement discord.py==2.2.2 (from octoprint-discordremote)ERROR: No matching distribution found for discord.py==2.2.2The update did not finish successfully. Please consult <code>octoprint.log</code> and <code>plugin_softwareupdate_console.log</code> for details.
```

# Oprint Logs
```
2023-06-04 01:33:47,592 - octoprint.plugins.softwareupdate - INFO - Starting update of discordremote to v4.6.0...
2023-06-04 01:33:59,467 - octoprint.plugins.softwareupdate - ERROR - Update of discordremote can not be performed, please also check plugin_softwareupdate_console.log for possible causes of this
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate/__init__.py", line 2193, in _perform_update
    "version": target_version,
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate/updaters/pip.py", line 127, in perform_update
    "Error while executing pip install", (stdout, stderr)
octoprint.plugins.softwareupdate.exceptions.UpdateError: Error while executing pip install
2023-06-04 01:33:59,628 - octoprint.plugins.softwareupdate - INFO - Saved update log to disk
2023-06-04 01:34:00,019 - octoprint.plugins.tracking - INFO - Sent tracking event update_failed, payload: {'target': 'discordremote', 'from_version': '3.11.0', 'to_version': 'v4.6.0'}
```

# pip testing

```
~/oprint/bin/pip list
Package                        Version
------------------------------ -----------
aiohttp                        3.7.4.post0
argon2-cffi                    21.3.0
argon2-cffi-bindings           21.2.0
asteval                        0.9.28
async-timeout                  3.0.1
attrs                          21.2.0
Babel                          2.12.1
blinker                        1.6.2
cachelib                       0.10.2
certifi                        2020.12.5
cffi                           1.15.1
chardet                        4.0.0
charset-normalizer             2.0.6
class-doc                      0.2.6
click                          8.1.3
colorlog                       6.7.0
commonmark                     0.9.1
continuousprint                2.4.0
Deprecated                     1.2.13
discord.py                     1.7.3
emoji                          2.4.0
feedparser                     6.0.10
filetype                       1.2.0
Flask                          2.2.5
Flask-Assets                   2.0
flask-babel                    3.1.0
Flask-Limiter                  3.3.1
Flask-Login                    0.6.2
frozendict                     2.3.8
future                         0.18.3
humanfriendly                  10.0
idna                           2.10
ifaddr                         0.1.7
immutabledict                  2.2.1
importlib-metadata             4.11.3
importlib-resources            5.12.0
itsdangerous                   2.1.2
Jinja2                         3.1.2
limits                         3.5.0
Markdown                       3.4.3
MarkupSafe                     2.1.1
more-itertools                 9.1.0
multidict                      5.2.0
netaddr                        0.8.0
netifaces                      0.11.0
OctoPrint                      1.9.0
OctoPrint-Dashboard            1.19.9
OctoPrint-DeleteMoveAfterPrint 1.10.0
OctoPrint-DiscordRemote        3.11.0
OctoPrint-FileCheck            2021.2.23
OctoPrint-FirmwareCheck        2021.10.11
OctoPrint-Fullscreen           0.0.6
OctoPrint-MalyanConnectionFix  0.1.3
OctoPrint-PiSupport            2023.5.24
OctoPrint-SibeBarTempGraph     0.1.6
ordered-set                    4.1.0
packaging                      21.3
passlib                        1.7.4
pathtools                      0.1.2
pathvalidate                   2.5.2
peerprint                      0.1.0
peewee                         3.15.4
Pillow                         8.4.0
pip                            20.3.3
pkg-resources                  0.0.0
pkginfo                        1.9.6
psutil                         5.9.5
pyasn1                         0.4.8
pycparser                      2.21
pydantic                       1.10.8
Pygments                       2.13.0
pylru                          1.2.1
pyparsing                      3.0.9
pyserial                       3.5
pysyncobj                      0.3.10
pytz                           2023.3
PyYAML                         5.4.1
regex                          2020.11.13
requests                       2.31.0
rich                           12.6.0
rsa                            4.0
sarge                          0.1.7.post1
semantic-version               2.10.0
sentry-sdk                     1.25.0
setuptools                     51.1.2
sgmllib3k                      1.0.0
six                            1.15.0
tornado                        6.2
typing-extensions              4.6.3
Unidecode                      0.4.21
urllib3                        2.0.2
watchdog                       2.3.1
webassets                      2.0
websocket-client               1.5.2
Werkzeug                       2.2.3
wheel                          0.36.2
wrapt                          1.15.0
yarl                           1.7.2
zeroconf                       0.39.4
zipfile2                       0.0.12
zipp                           3.8.0
zipstream-new                  1.1.8
zipstream-ng                   1.6.0
```

Anybody else having this issue? Thanks and lmk! If i find anything out ill attach it here.

cameroncros, 2023-06-04:
https://pypi.org/project/discord.py/2.2.2/

2.2.3 has been released, but 2.2.2 has not been removed. Try manually
installing 2.2.2 and then reinstalling the plugin? I am going to try an
entirely new solution for this problem shortly, but it could be a few weeks
away.

KenwoodFox, 2023-06-05:
Hey thanks for looking in on this with me!

```
~/oprint/bin/pip install discord.py==2.2.2
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: Ignored the following versions that require a different python version: 2.0.0 Requires-Python >=3.8.0; 2.0.1 Requires-Python >=3.8.0; 2.1.0 Requires-Python >=3.8.0; 2.1.1 Requires-Python >=3.8.0; 2.2.0 Requires-Python >=3.8.0; 2.2.1 Requires-Python >=3.8.0; 2.2.2 Requires-Python >=3.8.0; 2.2.3 Requires-Python >=3.8.0
ERROR: Could not find a version that satisfies the requirement discord.py==2.2.2 (from versions: 0.1.0, 0.2.0, 0.2.1, 0.3.0, 0.3.1, 0.4.0, 0.4.1, 0.5.0, 0.5.1, 0.6.0, 0.6.1, 0.6.2, 0.6.3, 0.7.0, 0.8.0, 0.9.0, 0.9.1, 0.9.2, 0.10.0, 0.11.0, 0.12.0, 0.13.0, 0.14.0, 0.14.1, 0.14.2, 0.14.3, 0.15.0, 0.15.1, 0.16.0, 0.16.1, 0.16.2, 0.16.3, 0.16.4, 0.16.5, 0.16.6, 0.16.7, 0.16.8, 0.16.9, 0.16.10, 0.16.11, 0.16.12, 1.0.0, 1.0.1, 1.1.0, 1.1.1, 1.2.0, 1.2.1, 1.2.2, 1.2.3, 1.2.4, 1.2.5, 1.3.0, 1.3.1, 1.3.2, 1.3.3, 1.3.4, 1.4.0, 1.4.1, 1.4.2, 1.5.0, 1.5.1, 1.6.0, 1.7.0, 1.7.1, 1.7.2, 1.7.3)
ERROR: No matching distribution found for discord.py==2.2.2
```

Weird, i can see that you linked directly to 2.2.2, but i don't see it! piwheels issue? I haven't touched the default octoprint sources witch should include both piwheels and regular pypi like you linked

JeremyVS, 2023-06-05:
I have also had this issue recently.  I ignored it at first (simply didn't update DiscordRemote), hoping a future update or DR would take care of it, but then DR disappeared from OctoPrint on me (not running/active, not in the list of installed plugins, and not in the Plugin Manager's list of things available to install, either!)

Installing with the v3.11.0.zip file, linked above, worked to bring it back... I get the error again if I try to let it update, but at least for now, DR is up and running again.  Will keep an eye out for updates, and can also try to provide logs if more data is needed, beyond what KenwoodFox has already given.

Thank you!

KenwoodFox, 2023-06-05:
> Wait, I know what your problem is, you have an old version of python. Try installing with: [`refs/tags/v3.11.0`.zip](https://github.com/cameroncros/OctoPrint-DiscordRemote/archive/refs/tags/v3.11.0.zip) I am also working on a new version without the python versioning issues, but that is still a few weeks off at least
> […](#)

I think i just have whatever version of python octoprint comes with, and I'm trying to stick to the stock setup because i want to like, not have incompatibilities like this one lol haha!

I can try installing from that zip sure, one sec

cameroncros, 2023-06-17:
@KenwoodFox If you are still interested in using this plugin, try this latest version of the plugin:

https://github.com/cameroncros/OctoPrint-DiscordRemote/archive/refs/heads/cloud-discordshim.zip

This is what the next version of DiscordRemote will be. Instead of setting up the discord bot yourself, everyone will just use my bot, which I am hosting on a server. There are no more python versioning issues with this arrangement, and the discord bot can be self hosted for those who care.

Code for the discord bot: https://github.com/cameroncros/discordshim_rs

KenwoodFox, 2023-06-24:
> @KenwoodFox If you are still interested in using this plugin, try this latest version of the plugin:
> 
> [`refs/heads/cloud-discordshim`.zip](https://github.com/cameroncros/OctoPrint-DiscordRemote/archive/refs/heads/cloud-discordshim.zip)
> 
> This is what the next version of DiscordRemote will be. Instead of setting up the discord bot yourself, everyone will just use my bot, which I am hosting on a server. There are no more python versioning issues with this arrangement, and the discord bot can be self hosted for those who care.
> 
> Code for the discord bot: [cameroncros/discordshim_rs](https://github.com/cameroncros/discordshim_rs)

I could set this up i suppose but, is the current version where each bot is hosted locally on the pi going away? if i could id just like to update my current version and keep it running as is. Not migrate to a whole new setup if at all possible.

cameroncros, 2023-06-25:
None of the old releases are going to go away, if what you have is working
fine there is no need to update. You will be able to roll back trivially as
well.

But setting up the bot is a really common sticking point for new people,
and this approach (if I can get it working) should make it easier.

It also decouples the discord.py version from the version of python
installed, which is hopefully helps the people on older octopi installs.

Right now, I am still having some issues, so don't upgrade unless you can
tolerate the bot being unreliable.

KenwoodFox, 2023-06-25:
I guess that makes sense, I'm a bot developer for other services so it didn't really affect me. I think I'd be down to self host the bridge. I just can't update the rest of my octoprint rn because it stops when it gets to discord

cameroncros, 2023-06-29:
5.0.0 is released, which should be a viable solution to this problem. I am going to close off this ticket.

closed 2023-06-29 by cameroncros