cameroncros/OctoPrint-DiscordRemote #253: V5.0 - Bot not working

cameroncros, 2023-06-29:
v5.0 has a completely new method of connecting to discord.

Your old bot from v2-4 will not be used anymore, and you will need to either:

- Setup discordshim: https://github.com/cameroncros/discordshim_rs

Or 

- Add the official discordshim bot to your server: https://discord.com/oauth2/authorize?client_id=433252064324354048&permissions=11264&scope=bot

CompuGenius, 2023-06-29:
@cameroncros I added the official discordshim bot to my server and filled in the channel ID, yet no test message or commands are working.

thenixhex311, 2023-06-30:
> @cameroncros I added the official discordshim bot to my server and filled in the channel ID, yet no test message or commands are working.

same here

cameroncros, 2023-06-30:
Does the discordshim bot have permission to post to the channel you have chosen?

CompuGenius, 2023-06-30:
> Does the discordshim bot have permission to post (and read messages) to the channel you have chosen?

Yes, of course

cameroncros, 2023-06-30:
I have restarted the discordbot, it was in an odd state, see if that helps. If not, can you contact me on discord to debug this further?

I am Cameron C on the octoprint discord.

CompuGenius, 2023-06-30:
> If not, can you contact me on discord to debug this further?
> 
> I am Cameron C on the octoprint discord.

Sent you a DM

thenixhex311, 2023-06-30:
I personally now had the "wake up ...what are we going to print today" appear, and now see "playing to 13 instances" under the bot on the userlist, but can't seem to get commands to work, going to see if it's something I messed up

MrFrickinFixit, 2023-07-01:
Same problem as well. I get a response from the bot after reinstalling the plugin, but nothing following that. No command responses or webcam snapshots.

cameroncros, 2023-07-05:
I think I have found the bug responsible for all your issues, and I will push a new version v5.1 shortly

CompuGenius, 2023-07-06:
Can confirm that the bot seems to work now!

MrFrickinFixit, 2023-07-08:
Can confirm v5.4 is working on all three of my Pi's.

MrFrickinFixit, 2023-07-08:
Ok, maybe not. Wake up messages work.  / commands work. Snapshots work. Start of print reports not working. Same for all three Pi's.

EvilGremlin, 2023-07-09:
now bot is ofline and don't respond to commands in channel, but manual commands from octo UI work

cameroncros, 2023-07-09:
There is some more bugs I need to fix, and my hosting provider has had some
outages overnight.

cameroncros, 2023-07-14:
Unless I hear otherwise, at this point everything seems stable and working? Please get in touch if you still have problems.

MrFrickinFixit, 2023-07-14:
Looks to be functioning correctly. Command and printer progress messages reporting as expected. Thank you for your efforts.

dshokouhi, 2023-07-25:
does the new bot respond to `/help` or anything? I cant get it to work after adding the bot and granting proper permission

cameroncros, 2023-07-25:
It should be responding, it appears to be up and working for me at least. Have you added the new bot to your discord and given it permission to read from the channel you want to send and recieve messages on?

dshokouhi, 2023-07-25:
> Have you added the new bot to your discord and given it permission to read from the channel you want to send and recieve messages on?

yup just double checked and those permissions are granted in teh channel the bot is added to

> @dshokouhi - Actually, it seems that it had stopped working for me, I have restarted the bot

still not responding to `/help` for me :(

cameroncros, 2023-07-25:
@dshokouhi Can you DM me on discord, I can try assist you. I am Cameron C on the Octoprint discord.

cameroncros, 2023-07-26:
Resolution seems to be removing and re-adding the bot to the discord server. Seems that permissions were incorrect somehow. (I recently got the bot verified, which may have impacted the permissions?

closed 2024-12-31 by cameroncros