cameroncros/OctoPrint-DiscordRemote #254: Be able to send mentions into the channel

Prototype7813, 2023-08-15:
**Is your feature request related to a problem? Please describe.**
Is it possible to make the @mentions work? I went to put my 3d printer thread in mute and received important parts only.

**Describe the solution you'd like**
Make the mentions work, if possible

cameroncros, 2023-08-16:
I haven't looked at this lately, but I believe that bots are intentionally
not allowed to @mention in embeds.

As things appear to be finally stable, I can look into @mentions again, but
I can't promise anything.

cameroncros, 2023-08-19:
Okay, I think mentions should now work. You'll need to add `<@433252064324354048>` to the message you want to send the mention. Replace the numbers with the user id of the person you want to mention. Let me know if there are any issues.

Prototype7813, 2023-08-20:
Hi Cameroncros! 

Is there something I need to do on my side to make it work? I just tried it and I have the same issue. 

Thanks for your fast response! 

[image]

Prototype7813, 2023-08-21:
Hi ! 

It's perfect ! It's working . The only thing is the mention appear in and out of the text area. But it's a minor thing . 

[image]

cameroncros, 2023-08-21:
Yup, that's intentional, unfortunately you can't have mentions inside the
embed. Annoyingly, it doesn't seem to do the substitution properly in the
title of the embed, so it looks a bit ugly.

Prototype7813, 2023-08-21:
At this point , it's working so I will not complain ! 

We can close this but maybe for improvement, it will be possible to add a field into the discord settings and add a checkbox for the message you want to add the mentions to and like that , it will hide the placeholder into the message.

Again thanks for the fast developpement

closed 2023-08-21 by Prototype7813