cameroncros/OctoPrint-DiscordRemote #257: Strange output of /status for Prusa XL

michalet, 2024-07-01:
**Describe the bug**
I have a Prusa XL 2T with connected Octoprint (1.10.2) with DiscordRemote (5.10.0). Print monitoring works fine, but the /status command returns more sensors than I would expect (screenshots below).
As my XL has two toolheads, I would expect readouts from them, possibly something from the unused slots. But the output shows six tools (the XL has max 5) and even more for extruders.

*Logs*
There is nothing in the discordremote log even when set to debug, but [here it is just in case](https://github.com/user-attachments/files/16057219/plugin_discordremote.log).

**To Reproduce**
Steps to reproduce the behavior:
1. Have a Prusa XL connected to Octoprint with DiscordRemote and Discord.
2. Use the /status command on discord.

**Expected behavior**
Only return readings from connected toolheads and bed. (Plus IPs, status and current Z).

**Screenshots**
[image]
[image]

cameroncros, 2024-07-02:
That is very interesting, this plugin just reports all the beds/extruders reported by octoprint. I'll have to investigate.

michalet, 2024-07-02:
I believe the B_X_X are for each tile of the bed. What is also intersting is that the temp graph in Octoprint itself correctly filters the two existing tools and bed.
jneilliii on Octoprint discord was kind enough to put together a regex filter to remove the extraneous readings, still have to test it to make sure it works though.
If I can provide some more info to help, let me know. ;)

cameroncros, 2024-07-02:
How does the regex work? Is it filtering the discord message?

cameroncros, 2024-07-02:
Okay, I think at best, all I can do is restrict the reporting to `toolX`, where X is 0,1,2,3,4,5, and report the `bed` value, and ignore all others. There is no way to know if the tools are valid or not, so you'd still have the extra tool2/3/4/5, but it might be a bit neater?

Alternatively, and I am kinda leaning this way anyway, I do nothing and your printer is a little bit noisier when reporting /status? Its not like the information is wrong, its just a bit meaningless for you specifically? I'm open to suggestions, but ideally I only want to make a general change, so that I don't end up special casing something for everyone.

cameroncros, 2024-07-03:
I've pushed an update, v5.11. It will let you specify the tools you wish to report in `/status`, see the new setting.

[image]

Can I resolve this issue?

michalet, 2024-07-04:
Yes, works perfectly. Thank you.

closed 2024-07-04 by michalet