cameroncros/OctoPrint-DiscordRemote #258: Host Discord Bot inside Octorprint

Spb2005, 2024-08-13:
Hello, 
i was wondering, if it was possible to host the discord bot inside the plugin, so while the printer/raspberry pi is online, the bot would be online an able to send messages. This would make the setup process very easy and you wouldnt need to host the bot on another computer or abroad. 

Another Idea i had, was able to get an automatic snapshoot after the first layer, which would be better than getting the snapshoot after completing 5% or so.

Thanks

cameroncros, 2024-08-14:
The previous version of the plugin did run the bot as part of the plugin. I ended up spending huge amounts of time responding to people who had issues with the discord bot setup. This new way is worse from a privacy standpoint, but i have far less issues reported.

It should be theoretically possible to install docker and build the discordshim container on your pi itself, but last time I tried it didn't work on armv7. I'll try again and report back.

Spb2005, 2024-08-14:
Thanks for the response, is the old version of the plugin still available somewhere and what improvements am i missing by using the old version with the bot inside octorprint.

cameroncros, 2024-08-14:
The armv7 branch seems to still be broken. if you need it, I can have another go at fixing it?

cameroncros, 2024-08-23:
In the absence of any updates, I will close this off now. Please re-open if you have any further issues

closed 2024-08-23 by cameroncros

twistedpixel, 2025-01-04:
Sorry to bump but I’m getting back into 3D printing and trying to revive my octoprint setup (with Discord remote) and it seems things have changed.

We all use your hosted bot now, correct? My only concern is that on the bot once added to my server, it says “serving 441 instances” or something to that effect. I’m guessing that’s how many people have added it. Does that mean all our messages are going through the same bot? Would that not present the possibility of image snapshots being shared? Or perhaps I misunderstand how these bots work.

cameroncros, 2025-01-04:
@twistedpixel 
You can self-host the bot, and I do suggest you do that if you are concerned about privacy.

The bot receives all messages on all channels it is added to. I don't log or store anything, but you kinda have to trust me on that, which is definitely not ideal if there is sensitive data.

To run your own not, see: 
https://github.com/cameroncros/OctoPrint-DiscordRemote/tree/v3.0.0?tab=readme-ov-file#create-the-discord-bot--in-discord
And:
https://github.com/cameroncros/discordshim_rs?tab=readme-ov-file#installation