cameroncros/OctoPrint-DiscordRemote #259: Setting hostname

Ponyboy47, 2025-03-22:
**Describe the bug**
The hostname is always displayed as `YOUR.HOST.NAME`

*Logs*

[plugin_discordremote.log](https://github.com/user-attachments/files/19403980/plugin_discordremote.log)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Octoprint
2. Click on 'Settings > DiscordRemote'
3. Scroll down to 'Show Local IP' & 'Show External IP'
4. Select the 'Hostname' option for both
5. Save settings
6. See message in Discord chat (example screenshot)

**Expected behavior**
Configured hostname should be displayed (e.g.: octoprint.example.com)

**Screenshots**

[image]

**Additional context**
It appears this value is the [hardcoded default](https://github.com/cameroncros/OctoPrint-DiscordRemote/blob/master/octoprint_discordremote/__init__.py#L230) and I can't find anywhere it would be overridden/saved.

cameroncros, 2025-03-23:
Weird that no one else has reported this yet :D. You can probably manually set it:

[image]

Longer term I will add a text box on the config page. Hopefully by the end of the week

cameroncros, 2025-03-24:
https://github.com/cameroncros/OctoPrint-DiscordRemote/releases/tag/v5.14.0

Should resolve your issue. However, there is now the issue that there is both baseURL and hostname in the settings, and I think there is a lot of overlap/potential for confusion there. Will leave it until someone complains though. 

Let me know if there are any further issues.

closed 2025-03-24 by cameroncros

Ponyboy47, 2025-03-25:
Thank you the latest update solved my need!